marcel-licence/ML_SynthTools #3: Initial error with precompiled library some ldflags thing

MichaelPNolan, 2022-01-17:
Ok - trying to start with an ESP32 and DAC
installed your (this) library manually

Can't compile because of this - and I just don't know anything about it - don't see any obvious advice to resolve. So would be a service if you can help me understand what is wrong? Version or something
Library ML_SynthTools has been declared precompiled:
Using precompiled library in /home/owner/Arduino/libraries/ML_SynthTools/src/esp32
The plaform does not support 'compiler.libraries.ldflags' for precompiled libraries.

MichaelPNolan, 2022-01-17:
Ok - i forked this and you updated so I resync-ed upstream and now the errors seem resolved. Having a rough couple of days making things work, broken hardware ... sigh.

closed 2022-01-17 by MichaelPNolan

marcel-licence, 2022-01-17:
I put a note into the readme file

MichaelPNolan, 2022-01-18:
Thanks for mentioning my fork ...
I've been building out (replacing most of the controls of an old CASIO) and
it has a midi serial out.
So i've got a good handle on the original midi serial stuff, which just
required the right parts.
I got the organ running with all your code (not the pro/fake organ yet). So
your code is easier and easier for me to get up and running and I
appreciate the developments. I'm super excited to make the "scope" work.

If I figure out what bug i have in my setup on Ubuntu linux Arduino IDE and
libraries ... i'll put that somewhere.
On Manjaro linux I can make
https://github.com/marcel-licence/ml_synth_organ_example work and
ML_SynthTools. So its obviously nothing wrong on your side.

I'm going to use the ADC module and breakout to do the 8 knobs with this
ml_synth_organ
Regards
Michael

marcel-licence, 2022-01-18:
Sounds great. Thank you for feedback.