marcel-licence/ML_SynthTools #24: Trying to port to TTGO T9 using WM8978 DAC

Aggebitter, 2022-12-18:
First of all, An impressive software. 
I have to boards from LilyGO that has all the hardware needed for this to run. I'm at the moment trying to get the TTGO T9 (TAudio) board to work ( https://github.com/LilyGO/TTGO-TAudio ).  Tried some sample code from (https://github.com/LilyGO/TTGO-TAudio/issues/12) and it runs. Then tried to remaster a new card following your code but I'm a bit confused with the nesting of "boards" vs DAC vs i2s_interface vs i2s_module. 

It might be better to separate hardware and its init ? 
Could you give it a try to add this board to the code ?

Lib common used for this DAC
https://github.com/CelliesProjects/wm8978-esp32
https://github.com/schreibfaul1/ESP32-audioI2S

marcel-licence, 2022-12-18:
Thank you.
The boards are just used as header files to have the pin definitions at a central point in it. I've added a prototype for the TTGO-TAudio some minutes ago. 

I agree that it would make sense to try to separate it somehow. I am looking forward to keep it more generic and flexible in future.

To use the WM8978 codec it might be just a short list of some configuration commands to get it running. 

I think you can try to include https://github.com/CelliesProjects/wm8978-esp32 only and use it to configure the DAC.
Regarding the I2S I could go another route. I tried to use one common interface for more than just ESP32 platforms.

Aggebitter, 2022-12-20:
_just started to try it out again without killing your code completely. Some initial issues. 
in file:  board_ttgo_t9_rgb_led_wm8978.h_

/**
 * @file board_ttgo_t9_rgb_led_wm8978.h  _<------------ typo change it ;)_
 * @author Marcel Licence
 * @date 20.12.2022 _<------------ and change the date_
 *
 * @brief Board description for TTGO-T9-RGB_LED-WM8978
 * @see https://github.com/LilyGO/TTGO-TAudio
 */

#ifndef BOARDS_BOARD_TTGO_T9_RGB_LED_WM8978_H_
#define BOARDS_BOARD_TTGO_T9_RGB_LED_WM8978_H_

/* on board led, green */
#define BLINK_LED_PIN     22 

#define MIDI_PORT2_ACTIVE
#define MIDI_RX2_PIN 34 /* use a free pin! */
#define RXD2 34  _<<< --  this is a strange work around I needed to get it to compile. I don't understand why the line above don't stick using your ml_synth_basic_example_

_In ml_synt_basic_example config.h I added the TTGO T9 in:_

//#define BOARD_ML_V1 /* activate this when using the ML PCB V1 */
//#define BOARD_ESP32_AUDIO_KIT_AC101 /* activate this when using the ESP32 Audio Kit v2.2 with the AC101 codec */
//#define BOARD_ESP32_AUDIO_KIT_ES8388 /* activate this when using the ESP32 Audio Kit v2.2 with the ES8388 codec */
//#define BOARD_ESP32_DOIT /* activate this when using the DOIT ESP32 DEVKIT V1 board */
#define BOARD_TTGO_T9_RGB_LED_WM8978 /* activate this when using the LilyGO TTGO T9 (Taudio) with WM8978 codec */
//#define BOARD_WEMOS_D1_MINI_ESP32
//#define BOARD_ADAFRUIT_QT_PY_ESPS2

_And future down_ 

#ifdef BOARD_ML_V1
#elif (defined BOARD_ESP32_AUDIO_KIT_AC101)
#elif (defined BOARD_ESP32_AUDIO_KIT_ES8388)
#elif (defined BOARD_ESP32_DOIT)
#elif (defined BOARD_TTGO_T9_RGB_LED_WM8978)

#define MIDI_PORT2_ACTIVE
#define MIDI_RX2_PIN RXD2

_Still trying to nest out the hardware init. It's a pain using arduino ide and my visual micro got some user rights issues when updating to arduino 2.03_ 

_I know that I should commit to your git instead of pushing it here but I haven't had the time to learn git._

Aggebitter, 2022-12-20:
got it running

closed 2022-12-20 by Aggebitter

marcel-licence, 2022-12-20:
Hi David,
thank you for providing your information. I will integrate it soon. I am currently trying to compile with the selected TTGO board. Hope I can remove the workaround and commit the update which will work in future.

marcel-licence, 2022-12-20:
I've updated the project and the header files.
In ml_boards.h the inclusion of the specific board header with specific define was missing. This may be causing the observed problems