marcel-licence/ML_SynthTools #25: ESP32 enhancement

copych, 2023-01-12:
Hello @marcel-licence, probably you've already seen this hint, which gives ~30% performance increase of float division according to my tests with Arduino IDE and ESP Arduino-SDK 2.0.6 . I thought, that this may help someday, so I'll leave it here for your attention.
https://blog.llandsmeer.com/tech/2021/04/08/esp32-s2-fpu.html

```
static __attribute__((always_inline)) inline float one_div(float a) {
    float result;
    asm volatile (
        "wfr f1, %1"          "\n\t"
        "recip0.s f0, f1"     "\n\t"
        "const.s f2, 1"       "\n\t"
        "msub.s f2, f1, f0"   "\n\t"
        "maddn.s f0, f0, f2"  "\n\t"
        "const.s f2, 1"       "\n\t"
        "msub.s f2, f1, f0"   "\n\t"
        "maddn.s f0, f0, f2"  "\n\t"
        "rfr %0, f0"          "\n\t"
        : "=r" (result)
        : "r" (a)
        : "f0","f1","f2"
    );
    return result;
}
```
and then replace 
`a = b / c;`
with
`a = b * one_div(c);`

PS float is not that precise (6-7 digits at best), you know. Funny thing is that both float division and this asm insertion accumulate errors in 5th significant digit already after 1000 continuous divisions, and the results differ.

marcel-licence, 2023-01-14:
Hello @copych 
thank you very much for this information.
I can try it in a benchmark soon.

In general the ESP32-S2 is not suited for floating point operations because the FPU is missing. In that case speeding up the division might not really help. Code for the ESP32-S2 should be designed like for the ESP8266 (avoiding using float). 
In addition to that you are right that the prevision could be better using other data types.

Thank you very much!

closed 2023-01-14 by marcel-licence

copych, 2023-01-14:
Please, get me right, this code is fpu instructions, and won't even compile for esp32-s2, this is only intended for esp32.

marcel-licence, 2023-01-15:
Ah okay. Do you know if the code works on the ESP32? I could imagine that in some cases the assembler statements are not compatible with the different cores. But from that perspective that would be great if it works =)

copych, 2023-01-15:
yes, the code works on ESP32, I have used it already.