marcel-licence/ML_SynthTools #29: Usage Question: RP2040 with PWM audio out

bradanlane, 2023-05-08:
I am assuming, to use a Raspberry Pi PICO and PWM output, I need to first set the defines for `ARDUINO_RASPBERRY_PI_PICO` and `RP2040_AUDIO_PWM`.

(1) What is not clear is the declaration of which GPIO pin(s) to use the code appears to use `0` and `1` but no mechanism to override them?

(2) Are two pins used for mono audio?

marcel-licence, 2023-05-08:
The pin is defined in 
https://github.com/marcel-licence/ML_SynthTools/blob/main/src/audio_module.h line 159.
I will update this to allow using an optional define outside of the library.

bradanlane, 2023-05-08:
Thanks for the quick reply. 

I'll likely fork the library as I only have mono PWM audio output.

marcel-licence, 2023-05-08:
You are welcome.

I've added `RP2040_AUDIO_PWM_PIN` as optional define to setup a specific start pin for stereo output. Stereo sound makes a big different especially for the multi saw synthesizer project.

bradanlane, 2023-05-08:
Appreciate the chew he and the recommendations. I'll revisit my HW design. 

When using the library with the RP2040, I am assuming I need to pick one of the PWM pairs:
[image]

marcel-licence, 2023-05-09:
I didn't test any other pin / pair yet. For efficient DMA usage it is recommended to use two PWM channels next to each other. In that case you should be able to use the library without any changes

closed 2023-05-09 by marcel-licence

bradanlane, 2023-05-10:
I tested several alternate pins for PWM.

Pin 0,1 work (default)
Pin 16,17 work

Pin 2,3 did not work
Pin 18,19 did not work
Pin 26,27 did not work

I can use pins as long as they result in PWM on Slice 0 (eg pins 0,1 and pins 16,17). any other slice does not produce sound output.

The code does compute the correct slice for other pins but for some reason, there is no output.

marcel-licence, 2023-05-10:
Seems that PWM0 A/B does work and others don't. It may be possible that something else is missing. Do you know if the IRQ was triggered?
Thank you very much for your investigation

bradanlane, 2023-05-10:
I don't have a hardware debugger configured (at the moment) so checking the IRQ will take some time to figure out.

bradanlane, 2023-05-10:
Can you suggest a way to TEST the IRQ without a hardware debugger ?