marcel-licence/ML_SynthTools #31: USB_MIDI_ENABLED vs MIDI_USB_ENABLED

bradanlane, 2023-05-10:
There are two definitions - `USB_MIDI_ENABLED` and `MIDI_USB_ENABLED` - which appear to be used similarly.

Should these all be one thing or could you explain when each is relevant?

marcel-licence, 2023-05-10:
I will remove USB_MIDI_ENABLED because it seems to have no function at all.
A document is prepared to explain useable defines: https://github.com/marcel-licence/ML_SynthTools/blob/main/extras/ml_defines.md

bradanlane, 2023-05-10:
From digging into the code, I see the following ...

#define USB_MIDI_ENABLED  // local enabled in ml_synth_multi_saw_example
#define MIDI_USB_ENABLED  // library enabled in ml_SynthTools

Enabling both generate errors compiling the ml_SynthTools

closed 2023-05-10 by marcel-licence

bradanlane, 2023-05-10:
`USB_MIDI_ENABLED` is used within library but `MIDI_USB_ENABLED` is used in most of the examples

bradanlane, 2023-05-10:
I guess `USB_MIDI_ENABLED` was never being set because the affected code in `midi_interface.h` has a typo.