marcel-licence/ML_SynthTools #35: ESP32-AudioKit ES8388 - I2C Init errors and no audio output

itarozzi, 2023-07-30:
I'm trying to execute EPiano and synth organ examples on my just arrived ESP32-AudioKit equipped with ES8388.

The board works with other libraries (arduino-audio-tool & arduino-audiokit), but I can't get working with this library.

I changed `board_audio_kit_es8388.h`  with following:
```
#define ES8388_CFG_I2C 2
#define ES8388_CFG_I2S 5
```

I also changed the config.h of example file to use Audiokit with ES8388:
```
#define BOARD_ESP32_AUDIO_KIT_ES8388 /* activate this when using the ESP32 Audio Kit v2.2 with the ES8388 codec */
```

At the boot, the serial console returns some errors about I2C CODEC initialization. I can play MIDI notes and they are printed in serial console, but no audio .

Here the boot log
```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
[  1104][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Loading data
Firmware started successfully
Initialize Audio Interface
Connect to ES8388 codec... [  2365][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=33 scl=32 freq=400000
0x00: 0x05
[  2470][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
0x01: 0x40
[  2483][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[  2487][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
STEP1[  2509][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
STEP2STEP3AdcCh1!
MixChAMPL!
[  2527][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[  2530][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
ES8388 setup finished!
Reg 0x00 = 0x05
[  2552][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x01 = 0x40
[  2568][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x02 = 0x00
[  2582][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x03 = 0x00
[  2596][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x04 = 0x3c
[  2610][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x05 = 0x00
[  2624][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x06 = 0x00
[  2637][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x07 = 0x7c
[  2651][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x08 = 0x00
[  2665][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x09 = 0x88
[  2679][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0a = 0x50
[  2693][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0b = 0x02
[  2707][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
Reg 0x0c = 0x0c
[  2721][E][Wire.cpp:416] beginTransmission(): Unfinished Repeated Start transaction! Expected requestFrom, not beginTransmission! Clearing...
[... 93 lines cut ...]
ESP.getMinFreeHeap() 174072
ESP.getHeapSize() 276752
ESP.getMaxAllocHeap() 110580
Total PSRAM: 0
Free PSRAM: 0
Firmware started successfully
<0x1b>[?25l<0x1b>[0;0HChorus_SetPhaseShift: 0.500     
901c64
>90 1c 64<
801c00
>80 1c 00<
901d64
>90 1d 64<
801d00
>80 1d 00<
901a64
>90 1a 64<
801a00
>80 1a 00<
901c64
>90 1c 64<
801c00
>80 1c 00<

```

marcel-licence, 2023-08-03:
It seems that the I2C configuration is correct but not sure about the I2S configuration. Could you ensure that the correct pins are used?
You could try to run the SAW test signal by defining: OUTPUT_SAW_TEST in config.h

itarozzi, 2023-08-04:
I enabled SAW test signal and it works - very loud :)

Then I recompiled without it and reconnected MIDI and now the EPiano example works.  The same with ml_synt_organ example. Really strange! 

But the list of errors from wire .cpp still remain. 

I will do more test tomorrow, but despite the errors it seems to work

marcel-licence, 2023-08-05:
Good to hear.
Strange. Maybe some component was not initialized correctly.
I don't know how to deal with the errors from wire. It may be possible that an end call is missing!?

Thank you very much for your feedback

itarozzi, 2023-08-05:
Can you address me to the file where the es8388 in the esp32 audiokit Is
initialized? I can do some debug here.

Thanks again

marcel-licence, 2023-08-05:
Yes sure: https://github.com/marcel-licence/ML_SynthTools/blob/main/src/es8388.h
You will find there 'ES8388_ReadReg' and 'ES8388_WriteReg'
Not sure if there is something wrong causing this error.

itarozzi, 2023-08-07:
May be I found the problem.

In es8388.h file, at the  ES8388_ReadReg function, you call `Wire.endTransmission(false);` twice.

I'm not much expert on i2c wire library, but watching some examples I think the last call is unnecessary.  Commenting out that line the error messages disappear.

Hope this helps.

```
uint8_t ES8388_ReadReg(uint8_t reg)
{
    Wire.beginTransmission(ES8388_ADDR);
    Wire.write(reg);
    Wire.endTransmission(false);

    uint8_t val = 0u;
    if (1 == Wire.requestFrom(uint16_t(ES8388_ADDR), uint8_t(1), true))
    {
        val = Wire.read();
    }
    // Wire.endTransmission(false);

    return val;
}

```

marcel-licence, 2023-08-07:
Thank you very much. Sounds good. Should Wire.endTransmission be called after the Wire.read instead of after calling it after Wire.write?

marcel-licence, 2023-08-07:
Hope this: https://github.com/marcel-licence/ML_SynthTools/commit/22ccbac55f664e05de9b3052a92252d232c5b589 does work

itarozzi, 2023-08-10:
Sorry for late feedback... holiday time :)

Unfortunately the 22ccbac commit doesn't work. The DAC initialization hangs.

You have to call `endTransmission` just after the last write, before to use `requestFrom`.

marcel-licence, 2023-08-10:
Thank you very much for your feedback. I've updated the code again moving the endTransmission call: https://github.com/marcel-licence/ML_SynthTools/commit/f1c9ecbaa72baab1bd8131fdb56ec44f13090250
I didn't had the time to prepare a setup to test it.

itarozzi, 2023-08-10:
Well done. Now it works fine. 

I think we can consider the issue closed.

Thanks again for sharing your project as free software

marcel-licence, 2023-08-10:
I am very happy to hear that.
New features and updates are in the queue.
Thank you for feedback and have a great holiday time =)

jordanlewis, 2024-07-28:
Hi @itarozzi, sorry for commenting on this old issue, but how did you both send MIDI over serial and read the serial console at the same time on the esp32-audiokit? I cannot figure out how to get that working - either Arduino is reading from the serial console, or I'm sending midi over serial (via the `serialmidi` program), but not both. Did you use 2 separate serial connections, somehow? Thanks!

itarozzi, 2024-07-28:
Hi @jordanlewis , I don't remember details of that Arduino sketch but from the  log attached to the first message I can see the MIDI use Serial2, while Serial is used for debug.

Often I use tty2midi on my Linux pc to send midi message to MCU so I can use a standard baud and a simple usb-ttl adapter.

marcel-licence, 2024-07-28:
I used this features with some own written tool which could display incoming data as text and could forward MIDI from another device to the serial port as raw. You should be able to run a second serial port on the ESP32 too, listening for MIDI messages on both connections. Hope this helps

jordanlewis, 2024-07-29:
Thank you, this is helpful!

For any future people with the same question, I used the program here: https://github.com/raspy135/serialmidi

and modified it to convert incoming bytes to ascii and just print them to the screen. That make it possible to send midi via serial, as well as viewing any logs from the esp32's serial output.

marcel-licence, 2024-07-30:
Did you try to upload the projects from the examples or did you try the project origin on GitHub? I observed some issues updating the submodules within this library