marcel-licence/ML_SynthTools #38: Support for ESP32S3 and ESP32 Arduino core 3.0.0?

nyh-workshop, 2023-11-19:
Hi Marcel Licence,

I have attempted to use this ML_SynthTools on the ESP32S3 board and could not get a correct sound output (Midi Stream Player) using the MAX98357A module. The output is mostly noise and it could be a wrong setting in the library.

It is suspected that the I2S drivers used are only for ESP32 and using the older ESP-IDF 4.x.

Any plans on porting this on the ESP32S3 and on newer versions of ESP32 Arduino core such as 3.0.0?

marcel-licence, 2023-11-19:
Hi,
I try to update the library over time. Unfortunately the ESP-IDF changes and that causes a lot of work to get the code updated in some cases.
Do you have an audio example? There are different types of problems you may face.
In addition to that I worked with the MAX some weeks ago. Another setting is required to get the output right. 
It may be just a small thing that should be changed to get it running.

Best regards,
Marcel

marcel-licence, 2023-11-19:
Did you insert
`#MAX_98357A_ENABLED `
in config.h?

nyh-workshop, 2023-11-20:
Hi @marcel-licence , thanks for the reply.

For the audio example, I'm using the `esp32_fm_synth` example from your repository, and disabled the midi serial modules such as `Midi_Setup()` and enabled the `MidiStreamPlayer_Init()` (trying to play a midi song with your FM synth). I have also added this `#MAX_98357A_ENABLED` into the example at the top of the sketch. However, I could only hear many noise coming and I suspect the I2S protocol might be not fitting the MAX98357A. (The MAX98357A modules I used are by Adafruit or its clones)

(btw the midi file is inside the flash - it can detect the file and plays it too)

Separately I tested my I2S startup with ESP-IDF 5.x (https://github.com/nyh-workshop/esp32s3-i2s-test) and this one works well without any problems. I compare this with the one in ML_SynthTool's i2s_interface.h and noticed that the MAX98357A implemented is `I2S_COMM_FORMAT_STAND_PCM_LONG` instead of the `I2S_COMM_FORMAT_STAND_I2S`.

I will try to connect this to the logic analyzer and try to use the `OUTPUT_SAW_TEST` to examine its output at the meantime. :D

marcel-licence, 2023-11-20:
Hi @nyh-workshop 
thank you for your feedback. Yes, it was intentional to use PCM_LONG:
```
#ifdef MAX_98357A_ENABLED
    .communication_format = I2S_COMM_FORMAT_STAND_PCM_LONG,
#else
    .communication_format = I2S_COMM_FORMAT_STAND_I2S,
#endif
```
see https://github.com/marcel-licence/ML_SynthTools/blob/main/src/i2s_interface.h line: 541

There is a pin SD_MODE which determines how to drive it:
https://www.analog.com/media/en/technical-documentation/data-sheets/max98357a-max98357b.pdf

I think I've connected SD via a resistor to VIN. It may happen that the channel settings is different.

nyh-workshop, 2023-11-21:
Hello @marcel-licence , thanks for the feedback. :D

Is it possible to add another option for the  normal basic standard I2S for the MAX98357A module? I used this a lot for most of my DACs and it has less wiring to do on the MAX98357A as it defaults to the standard I2S.

If it is not possible - I will perform this re-initialization outside the code for standard I2S in the main.ino file. :D

Also, I had compiled this in Arduino Core 3.0.0 alpha and it does not seem to crash or raise any exceptions - the only thing is the output is being noisy.

marcel-licence, 2023-11-21:
Hello @nyh-workshop ,
I would be happy to add some options. The lib should be easy to use.
I don't know if there a different MAX98357A breakout boards on the marked with different hardware settings.
You could try to fill one channel with zeros. Not sure what creates the noise on your end.
Do you have an audio example? There a different possibilities.

nyh-workshop, 2023-11-26:
Hi @marcel-licence ,
Apologies for the late reply. I had to modify the library for the ESP32S3 to fit into the newer ESP-IDF 5.x library (and the Arduino ESP32 core 3.0.0 alpha2), and its initialization routines are all separate from the ESP32. 

However, it does not currently support other "Audio In" routines and for now it is only focused on the ESP32 FM synth that you create. :D

After the modification, it is now playing a midi file in the flash (with the MidiStreamPlayer) - I'm not sure if the audio played is being correct or not. I'm getting a recording on this one now.

marcel-licence, 2023-12-23:
Hi @nyh-workshop do you have any updates on this? Not sure if there is anything left to do. I compiled the latest lib with ESP32 board library version 2.0.13. I think this should be independent from the Arduino core. But if it works, that would be nice =)

nyh-workshop, 2023-12-24:
Hi @marcel-licence , thanks for the help.

As I'm using this as a small midi player for a Christmas decoration, there are modifications with many ESP32S3 defines on this and it will be separate from the main Arduino core version 2.

I'm forking this with the ESP32S3 modifications for reference too. :D

closed 2023-12-24 by nyh-workshop

nyh-workshop, 2023-12-24:
The fork is created for ESP32S3: https://github.com/nyh-workshop/ML_SynthTools 

This works for ESP32 Arduino Core version 3. :D

marcel-licence, 2023-12-24:
Sounds great. Thank you very much for your feedback. Good to hear that it works. I am currently looking for a command line based format 1 to format 0 midi conversion tool. I would like to try to run the midi player in the sampler. I could also add the "still running" check. Maybe a callback could be nice to indicate when the playback has been finished. 
The version 3 is still in the alpha state? Seems there may be a lot of work coming in case 3.x will be release and people asking for compatibility.

nyh-workshop, 2023-12-29:
Hello @marcel-licence , thanks for the feedback and the improvements. :D 

From this [releases page](https://github.com/espressif/arduino-esp32/releases), the beta is still at 3.0.0 for ESP-IDF v5.0 support. On my ML_SynthTools fork, I have added temporary defines on the ESP32S3 to utilize the new ESP-IDF 5.0 libraries. It is not quite refined as I'm trying to get this to work earlier. :D

marcel-licence, 2024-01-13:
Hello @nyh-workshop thank you very much for your feedback. Do you know if it would be possible to determine during compilation if we are using v2.. or v3.. ? I would like to ensure that I have backwards compatibility until v3 is stable and out of its beta phase. Going from v1 to v2 took a lot of time.

nyh-workshop, 2024-01-14:
Hi @marcel-licence , I just had a search for determining the ESP-IDF version at code and this [page](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#c.ESP_IDF_VERSION_PATCH) explains how to do so. I have not tested this yet - do let me know if you encounter issues on this. :D