marcel-licence/ML_SynthTools #41: MidiStreamPlayer - function to check if the player is still playing music or not?

nyh-workshop, 2023-12-02:
Hello @marcel-licence ,

Is there a function or a way to check if the MIDI stream player is playing the MIDI file, or already stopped?

I examined the midi_stream_player.h on this and it looks like there are no functions to deal with the checking.

I'm actually writing a bit of a small MIDI player jukebox code and it could be great if the player could send a signal or to do something when the player is stopped. :D

nuigit, 2024-01-31:
Ah ... you have to predefine the new function as well. So add **bool MidiStreamPlayer_isPlaying();** before static bool midiPlaying = false; in the #ifdef ML_SYNTH_INLINE_DECLARATION section.

marcel-licence, 2024-01-31:
@nuigit thank you for your solution.
@nyh-workshop  I've included the function into the library. It should be available after an upgrade

closed 2024-01-31 by marcel-licence