marcel-licence/ML_SynthTools #42: Esp32s3 new sampler example

copych, 2023-12-24:
Hello, @marcel-licence, I have noticed the new promising example -- a sampler with sf support, but I couldn't compile it for ESP32S3, cause you probably haven't updated a compiled library under /src/esp32s3. Will you, please? ))

marcel-licence, 2023-12-25:
Hey, I've updated the lib today. Now you should be able to compile/link for the esp32s3 as-well

copych, 2023-12-25:
Thank you, I can confirm, I've managed to compile. Can you please advise, what is the easiest way to enable Hairless midi in for your example? I don't have standard 5-din midi by me right now, just a devBoard and a DAC )

marcel-licence, 2023-12-25:
Enabling '#define MIDI_RECV_FROM_SERIAL' should allow receiving MIDI messages via the serial port. It is used for all projects

copych, 2023-12-25:
Thanks and merry christmas!

closed 2023-12-25 by copych

marcel-licence, 2023-12-25:
Thanks, Merry Christmas!