marcel-licence/ML_SynthTools #44: missing predefines in midi_via_ble.h

nuigit, 2024-01-31:
Hello Marcel,
I got some compiler errors while trying the ble-midi-module. I think the predefinitions in midi_via_ble.h got lost somehow. By adding the definitions like this it worked like a charm.

_#ifdef ML_SYNTH_INLINE_DECLARATION
void midi_ble_setup();
void midi_ble_loop();
void Ble_rawMsg(uint8_t *msg);
void Ble_NoteOn(uint8_t ch, uint8_t note, uint8_t vel);
void Ble_NoteOff(uint8_t ch, uint8_t note);
void Ble_ControlChange(uint8_t ch, uint8_t number, uint8_t value);
void Ble_PitchBend(uint8_t ch, uint16_t bend);
void Ble_RttMsg(uint8_t msg);
void Ble_SongPos(uint16_t pos);
#endif /* ML_SYNTH_INLINE_DECLARATION */_

Best regards, Ron

marcel-licence, 2024-01-31:
@nuigit 
Hello Ron,
Thank you very much. I've added the missing definition.
Please let me know if the solution works.
Best regards,
Marcel

nuigit, 2024-02-01:
@marcel-licence 
Hello Marcel,
it's working like a charm, thanks for adding
Best regards,
Ron

closed 2024-04-18 by nuigit