marcel-licence/ML_SynthTools #45: to trully fix ESP32S3 I2S  interface...

matixan, 2024-04-09:
Here are some not usual #ifdef
https://github.com/marcel-licence/ML_SynthTools/blob/3b8cf78d514c704e08a32ad8390be5deb5457ff1/src/i2s_interface.h#L505-L507
change these to
```
#ifdef ARDUINO_RUNNING_CORE
#ifndef CONFIG_IDF_TARGET_ESP32S3
    .mck_io_num = I2S_PIN_NO_CHANGE,
#else
    .mck_io_num = I2S_MCLK_PIN,
#endif
#endif
```

Also
https://github.com/marcel-licence/ML_SynthTools/blob/3b8cf78d514c704e08a32ad8390be5deb5457ff1/src/i2s_interface.h#L531-L536

```#ifdef ES8388_ENABLED
	#ifndef CONFIG_IDF_TARGET_ESP32S3
    REG_WRITE(PIN_CTRL, 0xFFFFFFF0);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);
	#endif

CommentedNewline, 2024-06-28:
Why not write a PR?

marcel-licence, 2024-09-01:
I've updated the file. Thank you very much

closed 2024-09-01 by marcel-licence