marcel-licence/ML_SynthTools #46: platformIO issue

nuigit, 2024-04-18:
Hi Marcel, 
i tried to use your awesome library within vsCode and platformIO. 
I added https://github.com/marcel-licence/ML_SynthTools to lib_deps within 
platformio.ini but I get the following compiling error ... am I missing something?
 
... cannot find -lML_SynthTools

Best regards
Ron

marcel-licence, 2024-04-19:
Hi Ron,

I've created a branch for testing the platformIO integration: https://github.com/marcel-licence/ML_SynthTools/tree/platformIO
In that branch the .a files are renamed starting with lib...
In addition to that I've updated the readme with a statement to update the platform.ini:
https://github.com/marcel-licence/ML_SynthTools/tree/platformIO?tab=readme-ov-file#using-platformio
Please let me know if that solves the problem.

Best regards,
Marcel

nuigit, 2024-04-20:
Hi Marcel,
unfortunately it doesn't solve the issue. But maybe I'm doing something wrong. I tried implementing the library via _lib_deps = https://github.com/marcel-licence/ML_SynthTools.git#platformIO_ and also by copying the files directly in the lib-folder. Both times withe the provided build_flags. In main.h I just included "ml_config.h" ... no further code ... but still i get 

_/Users/ron.jagodzinski/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lML_SynthTools
collect2: error: ld returned 1 exit status
*** [.pio/build/esp32dev/firmware.elf] Error 1_

Is the build-flag provided within the library.json causing the issue?

Best regards,
Ron

marcel-licence, 2024-04-20:
Hi Ron,
I am sure we can figure this out. I tried using platformIO some days ago.
Did you check if "ML_SynthTools.a" was renamed to "libML_SynthTools.a"?
In addition to that in the platform.ini the path to the file is required. 

Because the library is mixed, some open source and some compiled stuff the flag -lML_SynthTools might be required to include all features.

I can retest this on my side too. My goal would be to have a description in the readme.md and a prepared lib for platformIO usage =)

Best regards,
Marcel

matixan, 2024-04-22:
You can use this script as temporary fix with master branch
``` python
# Description: This script is used to fix the issue of the missing library in the ML library. 
# The ML library is missing the lib prefix in the library name. This script will copy the library file to the correct location and rename it with the lib prefix.

import shutil
import os
Import("env")

print("Running ML fix action")

# Get path to the ML library which is located in lib
ml_lib_path = env.subst("$PROJECT_DIR") + "/lib/ML_SynthTools/src/esp32s3/"
ml_lib_name = "ML_SynthTools.a"
ml_lib_name_new = "lib" + ml_lib_name

print("Copying library file from: " + ml_lib_path + ml_lib_name + " to: " + ml_lib_path + ml_lib_name_new)

# copy the library file to the correct location and rename it with lib prefix
shutil.copy(ml_lib_path + ml_lib_name, ml_lib_path + ml_lib_name_new)

# done
```
and in .ini file put following
``` platformio
build_flags = 
    -L $PROJECT_DIR/lib/ML_SynthTools/src/esp32s3/

lib_deps = 
	ML_SynthTools

extra_scripts = 
    scripts/ML_lib_fix.py 
```

nuigit, 2024-04-23:
@matixan this seems to be working pretty well. Thanks a lot. 

I have some minor issues though. First I had to update the include path of _fs_ in ml_inline.h from #include _<fs\fs_something_ to #include _<fs/fs_something_ ... but maybe this is just mac specific. 
In addition I have some trouble with the implementation order in the ino files of the examples. Do you have a "known good" sample in platformio you could share?

Best regards
Ron

nuigit, 2024-04-23:
ok ... I got the my_synth_organ_example working with pio in vscode 
Thanks both of you for the support.
... would it be helpful ( by any means ) if I upload/send you the files?
Best regards
Ron

closed 2024-04-23 by nuigit