marcel-licence/ML_SynthTools #49: Add support for ESP32 board library v3.0.0

marcel-licence, 2024-06-02:
Version 3.0.0 is released and comes with some changes.
For example the usage of adc.h and i2s.h is deprecated and using it relies on legacy support.
A method is required to differentiate between 3.0.0 and earlier versions to use old and new code.

copych, 2024-10-31:
Hello @marcel-licence , if you're still looking for a method, there's `#if ESP_ARDUINO_VERSION_MAJOR < 3` directive, that works for me. Example here https://github.com/copych/ESP32-I2S-Audio-Helper-Class/tree/main/test_in_out

marcel-licence, 2024-10-31:
Hi  @copych thanks a lot. That helps to differentiate between 3.#.# and earlier versions. I got some sketches working with the new ESP32 version but some come with some unexpected errors.

marcel-licence, 2025-01-17:
Cannot be fixed until https://github.com/espressif/arduino-esp32/issues/9783 has been fixed

marcel-licence, 2025-01-17:
Please refer to https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html for migration