marcel-licence/ML_SynthTools #57: Failed compilation when using Midi Stream Player with RP2040/2350

nyh-workshop, 2024-10-03:
From git commit: #5656b69 and onwards

Hi Marcel Licence,
When I'm compiling a sample application [ML Synth Example](https://github.com/marcel-licence/ml_synth_fm_example) in Arduino, I encountered many compilation errors such as:

```
In file included from c:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/ml_inline.h:53,
                 from E:\pico-projects-2024\ml_synth_fm_example\ml_inline.ino:32:
c:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h: In function 'uint8_t MIDI_open(const char*, const char*)':
c:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:177:24: error: no matching function for call to 'fs::FS::begin(bool)'
  177 |     if (!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED))
      |          ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from C:\Users\yongh\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.1.1\libraries\LittleFS\src/LittleFS.h:29,
                 from E:\pico-projects-2024\ml_synth_fm_example\ml_synth_fm_example.ino:76:
C:\Users\yongh\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.1.1\cores\rp2040/FS.h:198:10: note: candidate: 'bool fs::FS::begin()'
  198 |     bool begin();
      |          ^~~~~
C:\Users\yongh\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.1.1\cores\rp2040/FS.h:198:10: note:   candidate expects 0 arguments, 1 provided
c:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h: In function 'void MidiStreamPlayer_ListFiles(uint8_t)':
c:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:545:32: error: no matching function for call to 'fs::FS::begin(bool)'
  545 |             if (!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED))
      |                  ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:\Users\yongh\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.1.1\cores\rp2040/FS.h:198:10: note: candidate: 'bool fs::FS::begin()'
  198 |     bool begin();
      |          ^~~~~
C:\Users\yongh\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.1.1\cores\rp2040/FS.h:198:10: note:   candidate expects 0 arguments, 1 provided
```

In the midi_stream_player it does not seem to contain any LittleFS specific includes for RP2040/2350. 

I have manually modified this midi_stream_player.h to include the LittleFS.h and it gives the same error, unfortunately.

Do you have examples of the midi stream player that is working for RP2040/2350? So far it works on the ESP32.

nyh-workshop, 2024-10-03:
There are even more undefined references if I took out the parameter in the LittleFs.begin. When I looked closely it seems that the begin for the RP2040/2350 do not take any parameters. So I tried removing this in the MIDI_open:
```
static uint8_t MIDI_open(const char *path, const char *mode)
{
    if (!LittleFS.begin())
    {
        Serial.println("LITTLEFS Mount Failed");
        return 0;
    }
```

Compile Output:
```
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\AppData\Local\Temp\arduino_build_440122\sketch\ml_synth_fm_example.ino.cpp.o: in function `_Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch':
C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:356:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x38): undefined reference to `_Z21midi_file_stream_loadPcP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:360:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x50): undefined reference to `_Z16interpret_uint16Ph'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:376:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x64): undefined reference to `_Z26MidiStreamReadTrackPrepareP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:380:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x76): undefined reference to `_Z29MidiStreamReadSingleEventTimeP11midi_proc_sPl'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:364:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x86): undefined reference to `_Z20MidiStreamParseTrackP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:372:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0xc6): undefined reference to `_Z19MidiStreamSkipTrackP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\AppData\Local\Temp\arduino_build_440122\sketch\ml_synth_fm_example.ino.cpp.o: in function `_Z21MidiStreamPlayer_Tickm':
C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:504:(.text._Z21MidiStreamPlayer_Tickm+0xe): undefined reference to `_Z16interpret_uint16Ph'
```

marcel-licence, 2024-10-03:
Hello, thank you very much for your message. I will check this soon. I've created this fs/... files to be able to have one interface with different ports for the different platforms. I think I should update the stream player to use this interface to avoid a bunch of changes in case a platform changes.

marcel-licence, 2024-10-03:
I've updated the implementation now using the fs implementation which might be a bit more future proof. The rp2350 implementation needs some more work because I got some trouble with linking some libs. The new rp2040 board library works a bit different and this causes trouble with some code.
libML_SynthTools_RP2040.a -> must be moved one folder up to be found
This might break other projects accidentality try to link code from the lib.

nyh-workshop, 2024-10-06:
Hello Marcel Licence, thanks for the update. The compilation works for the midi stream player. The errors are now reporting the missing functions that are not compiled for the RP2350 libraries yet.

```
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\AppData\Local\Temp\arduino_build_610523\sketch\ml_synth_fm_example.ino.cpp.o: in function `_Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch':
C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:232:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x38): undefined reference to `_Z21midi_file_stream_loadPcP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:236:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x50): undefined reference to `_Z16interpret_uint16Ph'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:252:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x64): undefined reference to `_Z26MidiStreamReadTrackPrepareP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:256:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x76): undefined reference to `_Z29MidiStreamReadSingleEventTimeP11midi_proc_sPl'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:240:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0x86): undefined reference to `_Z20MidiStreamParseTrackP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:248:(.text._Z42MidiStreamPlayer_PlayMidiFile_fromLittleFSPch+0xc6): undefined reference to `_Z19MidiStreamSkipTrackP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\AppData\Local\Temp\arduino_build_610523\sketch\ml_synth_fm_example.ino.cpp.o: in function `_Z21MidiStreamPlayer_Tickm':
C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:380:(.text._Z21MidiStreamPlayer_Tickm+0xe): undefined reference to `_Z16interpret_uint16Ph'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:388:(.text._Z21MidiStreamPlayer_Tickm+0x5a): undefined reference to `_Z25MidiStreamReadSingleEventP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:391:(.text._Z21MidiStreamPlayer_Tickm+0x68): undefined reference to `_Z29MidiStreamReadSingleEventTimeP11midi_proc_sPl'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:368:(.text._Z21MidiStreamPlayer_Tickm+0x9e): undefined reference to `_Z16MidiStreamRewindP11midi_proc_s'
C:/Users/yongh/AppData/Local/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/midi_stream_player.h:370:(.text._Z21MidiStreamPlayer_Tickm+0xa6): undefined reference to `_Z29MidiStreamReadSingleEventTimeP11midi_proc_sPl'
collect2.exe: error: ld returned 1 exit status

```

marcel-licence, 2024-10-06:
Hello, thank you for your feedback.
I am working on it.

marcel-licence, 2024-10-15:
Hey, I've updated the lib including a separation into ML_SynthTools and ML_SynthTools_Lib.
Would you like to update both and recompile again?

nyh-workshop, 2024-10-15:
Hello @marcel-licence , thanks for the update too. :D 
I have already git pull and recompiled it - this time I get another new compile error:
```
Alternatives for librp2350_i2s_audio.h: []
ResolveLibrary(librp2350_i2s_audio.h)
  -> candidates: []
In file included from C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/ml_inline.h:39,
                 from E:\pico-projects-2024\ml_synth_fm_example\ml_inline.ino:32:
C:\Users\yongh\OneDrive\Documents\Arduino\libraries\ML_SynthTools\src/audio_module.h:128:10: fatal error: librp2350_i2s_audio.h: No such file or directory
  128 | #include <librp2350_i2s_audio.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

I think the rest of the libraries are already found and compiled.

marcel-licence, 2024-10-15:
Seems something is missing. The file is included into the lib:
https://github.com/marcel-licence/ML_SynthTools_Lib/blob/main/src/librp2350_i2s_audio.h

nyh-workshop, 2024-10-16:
Hi @marcel-licence , thanks for the response.

From your github seems that you have separated this repository into another [library](https://github.com/marcel-licence/ML_SynthTools_Lib/tree/main/src)?

Will there be a submodule, or I should download this SynthTools_Lib separately and manually install it back into the ML_SynthTools? :D

marcel-licence, 2024-10-16:
Hey, this needs to be downloaded/installed separately. Hope I can prepare some .json for better Arduino integration soon. A submodule seems not to be supported within the Arduino IDE

nyh-workshop, 2024-10-17:
Hi @marcel-licence , thanks for the help again. :D

This time using the latest commit for ML_SynthTools `04c06d5` and the library `1bace70` I managed to get it to compile successfully too.

closed 2024-10-17 by nyh-workshop