marcel-licence/ML_SynthTools #64: FM synth example - do we have preset algorithms?

nyh-workshop, 2024-10-17:
I have just compiled the FM synth example with the latest ML_SynthTools build [04c06d5](https://github.com/marcel-licence/ML_SynthTools/commit/04c06d5e86ca6aa05a2243fb460564eb058b6b8d) and it looks like I need a external MIDI controller to manually create the algorithms.

Just curious - is it possible to have default preset algorithms like DX7/9's "E.Piano" and such for one to start inside, instead of having to set them manually with an external MIDI device? Having this to run as a midi player box is one of the ideas I have. :D

marcel-licence, 2024-10-17:
Hi,
I tried to make some presets placed on the different channels in https://github.com/marcel-licence/ML_SynthTools/blob/main/src/ml_fm.cpp line 419ff.

I didn't emulate a real device which might work with integer values instead of float. It could be possible to try to adopt settings from the devices.

This implementation does not have the same complexity like the DX7. I need to add some more operators to get the similar DX7 typical sound out of it

marcel-licence, 2024-10-17:
I might miss some point. Regarding the internal algorithms the implementation is the same as shown in this video: https://youtu.be/rGTw05GKwvU
I just cleaned up my repositories and added more support.