marcel-licence/ML_SynthTools #69: Sampler example - sample buffer panic

xdissent, 2024-10-31:
I've been trying to get the `ml_synth_sampler_example` working but I always get a `LoadProhibited` panic when accessing the sample buffer. I get the same error on esp32-wrover, esp32-s3, and esp32-audiokit-es8388. All have psram enabled.

The only change I've made aside from config for different boards is:

```
-    //Sampler_NoteOn(0, 69, 127);
+    SoundFontSamplerCtrl(1);
+    Sampler_NoteOn(0, 69, 127);
```

I've put the `PappRohrSample.wav` on littlefs and it appears to be loaded correctly:
```
LittleFS Mount Successful!
List files ...

PappRohrSample.wav              15334
setup done!
SoundFontSamplerCtrl: 1
Reading wav: /PappRohrSample.wav
  fileSize: 15326
  lengthOfData: 16
  byteRate: 96000
  bytesPerSample: 2
  bitsPerSample: 16
  sampleRate: 48000
  numberOfChannels: 1
data found
error reading tag
Sampler - Samplecount: 1
Sampler - Bufferusage: 0.370
```

Then whenever I do `Sampler_NoteOn()` via code or midi message I get:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8526  PS      : 0x00060830  A0      : 0x800d3b14  A1      : 0x3ffb1f30  
A2      : 0x00010000  A3      : 0x0007ffff  A4      : 0x0000007f  A5      : 0x00010000  
A6      : 0xffffffff  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1fd0  
A10     : 0x3ffb1f70  A11     : 0x3ffd15b0  A12     : 0x00000030  A13     : 0x3ffcdd58  
A14     : 0x3ffd17bc  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400d8514  LEND    : 0x400d85e0  LCOUNT  : 0x0000002f  

Backtrace: 0x400d8523:0x3ffb1f30 0x400d3b11:0x3ffb1f60 0x400de3f9:0x3ffb2290
```

I'm compiling (for audio kit) with `arduino-cli compile -v -b esp32:esp32:esp32:PSRAM=enabled,PartitionScheme=no_ota .`
The libs used are:
```
Used library            Version Path
ML SynthTools           2.0.0   /Users/xdissent/Documents/Arduino/libraries/ML_SynthTools
Wire                    2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/Wire
SPI                     2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/SPI
ML SynthTools Libraries 2.0.0   /Users/xdissent/Documents/Arduino/libraries/ML_SynthTools_Libraries
WiFi                    2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/WiFi
FS                      2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/FS
LittleFS                2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/LittleFS
SD_MMC                  2.0.0   /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/libraries/SD_MMC

Used platform Version Path
esp32:esp32   2.0.13  /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13
```

ML SynthTools and ML SynthTools Libraries are the latest versions from git.

I can successfully play a sample using `Sampler_NewSampleStatic()`, but I don't believe that loads it into the sample buffer like wav/sf samples. I'm assuming it's some kind of buffer access issue? I get the same error when I try to use `Sampler_UseStaticBuffer()`

I've been able to load/play a wav sample with a pico 2 using static buffer, but the sound is just noise. Not sure what that problem is but it makes me think my sample buffer issue is esp32-specific.

Any ideas? Thanks!

xdissent, 2024-10-31:
Yes that's what I thought... but the sample buffer is clearly not null:

```
    static uint8_t *storage = (uint8_t *)ps_calloc(storageBytes, 1);
    if (storage == NULL)
    {
        Serial.printf("Not able to allocate the complete PSRAM buffer!\nNow trying to reduce the allocation buffer size...\n");

        /*
         * for some reason using a newer ESP32 board library you cannot allocate the complete PSRAM memory
         * this loop will decrease the buffer size and repeat the allocation step again until it is successful
         */
        while ((storage == NULL))
        {
            storageBytes -= 1;
            storage = (uint8_t *)ps_calloc(storageBytes, 1);
        }
        Serial.printf("alloced %" PRIu32 " bytes\n", storageBytes);
    }

    Sampler_SetSampleBuffer(storage, storageBytes);
```

Output:
```
Try to allocate 4192123 bytes
Not able to allocate the complete PSRAM buffer!
Now trying to reduce the allocation buffer size...
alloced 4128756 bytes
```

So it's def making it thru the `while ((storage == NULL))` loop. I'm wondering if `Sampler_SetSampleBuffer()` is using the wrong pointer or something, or if there's some scope issue I don't understand.

copych, 2024-10-31:
If you don't have EspExceptionDecoder, you should get one (AFAIK it only works in Arduino IDE 1.x)
https://github.com/me-no-dev/EspExceptionDecoder this would let localize the exception source.

marcel-licence, 2024-10-31:
Hi @xdissent regarding the compilation. That looks okay so far. I will compile and test it too

xdissent, 2024-10-31:
Great tip @copych!

```
Decoding stack results
0x400d3b15: loop() at /Users/xdissent/Code/ml_synth_sampler_example/ml_synth_sampler_example.ino:483
0x400de3e1: loopTask(void*) at /Users/xdissent/Library/Arduino15/packages/esp32/hardware/esp32/2.0.13/cores/esp32/main.cpp:50
```

That line is `Sampler_Process(left, right, SAMPLE_BUFFER_SIZE);` 🤷‍♀️ 

I don't know if there's a way to get backtraces further into the precompiled sampler lib.

marcel-licence, 2024-10-31:
The static sample feature is new and plays it completely from flash without using the RAM. Maybe I created some issue by adding this feature.

marcel-licence, 2024-10-31:
Seems I found the bug. I can push an update in a few minutes

marcel-licence, 2024-10-31:
I've commited an update to ML_SynthTools_Lib. Please try this one

xdissent, 2024-10-31:
Success!!! Thank you so much!

I did have to remove `ML_SynthTools.a` so my compiler would pick up the updated `libML_SynthTools.a` instead. But I can load and play the sample with no crash!

closed 2024-10-31 by xdissent

marcel-licence, 2024-10-31:
@xdissent  Did you get an error that something was defined twice? In that case I will remove the other .a file.

xdissent, 2024-10-31:
I didn't get a build error, but the compilation output explicitly mentioned it was linking against `ML_SynthTools.a` and my sketch still had the same panic when I uploaded. Then I deleted it and the compilation output didn't mention the lib explicitly, just `-lML_SynthTools` and it seemed to find `libML_SynthTools.a` correctly. I don't know why it would choose one over the other.

marcel-licence, 2024-10-31:
My fault, did forget to update the other file. But I removed it because it has no use any more. I was hoping to fix the esp32 v3 build but this needs some other changes. Thank you very much for your feedback.

xdissent, 2024-10-31:
Hey thank *you* very much! This thing is so much fun! Happy Halloween! 🎃