marcel-licence/ML_SynthTools #81: invalid conversion from 'int' to 'Stream*'

renatoni, 2025-02-09:
Ich wollte das Beispielprojekt mit ESP32 ml_synth_organ_example mal zum Laufen bringen, scheitere aber schon beim ersten kompilieren mit Fehlern.
Die ML_SynthTools habe ich in die ArduinoIDE installiert.
Das Kompilieren bricht mit folgenden Fehlern ab:

In file included from C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\ml_synth_organ_example.ino:49:
C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\ml_synth_organ_example.ino: In function 'void setup()':
C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\config.h:159:21: error: invalid conversion from 'int' to 'Stream*' [-fpermissive]
  159 | #define SAMPLE_RATE 44100
      |                     ^~~~~
      |                     |
      |                     int
C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\ml_synth_organ_example.ino:178:17: note: in expansion of macro 'SAMPLE_RATE'
  178 |     Organ_Setup(SAMPLE_RATE);
      |                 ^~~~~~~~~~~
C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\ml_synth_organ_example.ino:178:16: error: too few arguments to function 'void Organ_Setup(Stream*, float)'
  178 |     Organ_Setup(SAMPLE_RATE);
      |     ~~~~~~~~~~~^~~~~~~~~~~~~
In file included from C:\Users\renato.nitsch\Documents\Arduino\ml_synth_organ_example\ml_synth_organ_example.ino:60:
c:\Users\renato.nitsch\Documents\Arduino\libraries\ML_SynthTools-main\src/ml_organ.h:84:6: note: declared here
   84 | void Organ_Setup(Stream *ser, float sample_rate);
      |      ^~~~~~~~~~~

exit status 1

Compilation error: invalid conversion from 'int' to 'Stream*' [-fpermissive]

Hoffentlich kann mir geholfen werden. Ich komme da einfach nicht weiter.

Herzliche Grüße

marcel-licence, 2025-02-11:
Hallo @renatoni ,
vielen Dank für Deine Nachricht. Ich freue mich immer sehr über das positive Feedback.

Hier gibt es einen Konflikt mit der Version von dem Projekt und der Version von den ML_SynthTools.

`void Organ_Setup(Stream*, float)` ist eine recht veraltete Deklaration. Ein Update sollte das Problem beheben können.

Viele Grüße
Marcel

renatoni, 2025-02-13:
Hallo Marcel,
die ML_SynthTools hatte ich erst die Tage von hier: 
https://github.com/marcel-licence/ML_SynthTools  
heruntergeladen.
Das Projekt "ml_synth_organ_example" hatte ich mir ebenfalls die Tage von hier:
https://github.com/marcel-licence/ml_synth_organ_example
geladen.
Interessant ist, dass in der herunter geladenen zip von ML_SynthTools, die Beispielordner zum Teil leer sind. So eben auch das  ml_synth_organ_example Beispiel. Daher habe ich das Projekt auf die andere Weise über den angegebenen Link geladen.

Viele Grüße
Renato